adlerre/obd2-mqtt #8: Error using ESPHome

marcoraddatz, 2024-10-15:
Hi, the device arrived today so I gave it a try.

The detection was smoothly and the install was straight forward. However, the device currently in a dead loop, this is the message I get (multiple times):

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (765) esp_core_dump_flash: No core dump��ѥѥ���found!
E (765) esp_core_dump_flash: No core dump partition found!
Start Access Point...WiFi AP stopped.
WiFi AP started.
AP - IP address: 192.168.4.1
HTTP server started
Start modem...
Initializing modem...
Failed to restart modem, delaying 10s and retrying
WiFi AP stopped.
ets Jul 29 2019 12:21:46
```

Do the partition tables get flashed via ESPHome or do I need to do it manually first? Must I already have a working SIM for the setup process?

Thanks,
Marco

adlerre, 2024-10-15:
Hello Marco,

If you use the web installer, all the things you need will be flashed onto the device. However, the problem seems to be the wrong device selection. The device type should be printed on the circuit board. For an A7670, this can be V 1.0 or 1.1.

To get the whole thing to work, a SIM card and ELM327 adapter are essential.

René

adlerre, 2024-10-15:
I tried to reproduce the error and encountered the same error when no SIM card was inserted. Perhaps a hint should be given here.

The error with the partition is known and intended, otherwise 64k would be missing for OTA updates.

Since there are known problems using WiFi and Bluetooth at the same time (since both use 2.4GHz), it can take a while for the AP to appear.

marcoraddatz, 2024-10-15:
> I tried to reproduce the error and encountered the same error when no SIM card was inserted. Perhaps a hint should be given here.

Would create a PR with an updated readme if it helps.

> The error with the partition is known and intended, otherwise 64k would be missing for OTA updates.

Good to know.

> Since there are known problems using WiFi and Bluetooth at the same time (since both use 2.4GHz), it can take a while for the AP to appear.

The AP is now always available, but for whatever reason I currently cannot access it via browser. 

How is the connection supposed to work in general: Do you always rely on the LTE data transfer? Or might it be an option to set priorities? E.g. Wi-Fi is available, use this. Otherwise use cellular data? 
I'm asking because if the connection via Wi-Fi would be established within the setup process and the cellular connection would be optional until a SIM gets plugged in, it would make the whole process easier for new users.

adlerre, 2024-10-15:
> > I tried to reproduce the error and encountered the same error when no SIM card was inserted. Perhaps a hint should be given here.
> 
> Would create a PR with an updated readme if it helps.
> 
Would be nice.
> Good to know.
>
>
> The AP is now always available, but for whatever reason I currently cannot access it via browser.
It might help to set an IP address on the client itself, i.e. 192.168.4.2.
> How is the connection supposed to work in general: Do you always rely on the LTE data transfer? Or might it be an option to set priorities? E.g. Wi-Fi is available, use this. Otherwise use cellular [...]

This is because it is trying to establish a Bluetooth connection. I will counteract this by interrupting the scanning as soon as a connection to the AP is established.

The idea is that an LTE connection must always exist, since most cars do not have their own WiFi AP.

marcoraddatz, 2024-10-15:
> It might help to set an IP address on the client itself, i.e. 192.168.4.2.

Did that, but will try again once I install the ELM327.

> The idea is that an LTE connection must always exist, since most cars do not have their own WiFi AP.

How unfortunate! 😄 This was exactly what I was aiming at: my car already has its own cellular connection and would be able to provide a Wi-Fi access point all the time. Now I'm thinking about reversing it and to move the SIM to the ESP32 to spin up an AP where my other devices can connect to.

adlerre, 2024-10-15:
> How unfortunate! 😄 This was exactly what I was aiming at: my car already has its own cellular connection and would be able to provide a Wi-Fi access point all the time. Now I'm thinking about reversing it (somehow) and to move the SIM to the A7670E to spin up an AP where my other devices can connect to (if easily possible).
> 
> You disable the AP because of connectivity issues with Bluetooth, right?

This is not possible because if there is a connection to the AP, the Bluetooth connection is interrupted.
https://github.com/adlerre/obd2-mqtt/blob/47b4e9a4afb186949e238b40b429f705284eae73/src/main.cpp#L116

With arduino framework WiFi/Bluetooth coexistence isn't working. It should be possible with the [espidf framework](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/coexist.html), but most of the used libraries wouldn't work.

marcoraddatz, 2024-10-16:
Thanks, I'm closing this one. I'll continue with the setup in a few days. If I find out why the web page can't be accessed, I'll open a new issue instead.

closed 2024-10-16 by marcoraddatz

aleks-mariusz, 2024-10-20:
@adlerre great project btw! looking forward to giving this one a try

One thing:

> device type should be printed on the circuit board

Can you make this clearer on the web-installer? I spent some time scratching my head figuring out what the diff variations of the A76xxx were, which could have been avoided if i knew this key bit of info :-)

adlerre, 2024-10-21:
> Can you make this clearer on the web-installer? I spent some time scratching my head figuring out what the diff variations of the A76xxx were, which could have been avoided if i knew this key bit of info :-)

It's done with #14