adlerre/obd2-mqtt #10: Connection via WiFi instead of cellular

terdt, 2024-10-17:
Hi René, if understood it correct the MQTT messages will be sent via GSM or LTE to the broker. Is there also a possibility that I can use a normal ESP without cellular modem ? The ESP should connect via WiFi to an Hotspot / AP.  Background ist that I plan to use it an RV (Wohnmobil) where an WiFi Network already exists. Kind Regards Thomas

adlerre, 2024-10-17:
Hi Thomas, 
this isn't possible because Bluetooth and WiFi shares the same frequencies. ESPs used with arduino framework does not realy work with coexistence. This should possible with the [espidf](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/coexist.html) framework, but the most of used libraries wouldn't work with this.

René

terdt, 2024-10-17:
Hi René, thanks for fast response and clarification.

closed 2024-10-17 by terdt