adlerre/obd2-mqtt #17: Add visual input validation

marcoraddatz, 2024-10-22:
Right now, the inputs are validated, but it's not clear, why the "save" button is disabled. I think it might be a good and relatively easy feature to show validation errors (e.g. "PIN is not a number", "APN is required", etc.).

Also, it might be good to visually highlight the required inputs with an asterisks. 

[image]

adlerre, 2024-10-22:
It sounds a little bit weird, but want save some space and don't use error descriptions.
The UI needs currently 140k and the partition size is only 196k.
But now required fields marked directly.

closed 2024-10-22 by adlerre

marcoraddatz, 2024-10-22:
What about using HTMX instead of Angular? From my experience, this would shrink the size to less than 100k.

adlerre, 2024-10-22:
Because i‘m love Angular. 
Was der Bauer nicht kennt… 😆