adlerre/obd2-mqtt #20: Enhancement idea: Wireguard support

spitfire, 2024-10-23:
Hi,

I'd rather not expose my MQTT broker to the internet, but I do have Wireguard server installed on my router. I know ESPHome has a component to connect using that, so it should be possible on esp32 devices. would you consider adding that?

adlerre, 2024-10-24:
Hi,
sorry, but i don't want to support any kind of VPN. If you don't want to expose your broker to the internet, use a free MQTT provider and subscribe to the topic with your HA.

René

closed 2024-10-24 by adlerre

spitfire, 2024-10-24:
Can I do that even when I'm already using an MQTT broker installed on my system?

adlerre, 2024-10-24:
Not directly! But it's thinkable to build a MQTT-Proxy-Service, with a connection to the public server witch push the topics to the HA MQTT.
This should easily done with Node and [MQTT.js](https://github.com/mqttjs/MQTT.js).

adlerre, 2024-10-24:
Hope that's help -> #21

spitfire, 2024-10-24:
thanks, I'll take a look at it!