adlerre/obd2-mqtt #25: No values ​​from car VW e-UP

sstuercke, 2024-11-12:
Hi René,
the connection to the MQTT broker works.
However, no values ​​are read on my VW e-UP.
Is the VW e-Up not supported?

[VWeUP.txt](https://github.com/user-attachments/files/17721724/VWeUP.txt)

adlerre, 2024-11-13:
Hi,

I can't say whether it is supported or not. Because I've never tested it with an EV.

If you want to help me, you can check the box for "Check if PID is supported" in the settings and test it again on your car. After that, I'm interested in the values ​​for supportedPids_*.

sstuercke, 2024-11-14:
Hi René, 
here are the supportedPids.
13e02f8d5d6d_supportedPids_1_20 = 10000000000010000000000000010001
13e02f8d5d6d_supportedPids_21_40 = 10000000000010000000000000010001
13e02f8d5d6d_supportedPids_41_60 = 1110000010000000000000000000000000
13e02f8d5d6d_supportedPids_61_80 = 11000100110000111000000000100001

Of interest is the SoC and whether it is connected to the wallbox.
I have also enclosed a protocol.
[esp-web-tools-logs.txt](https://github.com/user-attachments/files/17749123/esp-web-tools-logs.txt)
[VWeUP.txt](https://github.com/user-attachments/files/17749126/VWeUP.txt)

Thanks

adlerre, 2024-11-14:
Thank you, that helps me a lot. I'm analyzing it and trying to implement support. But please be patient, as I'm currently working on a major rebuild.

But I could provide a branch for testing in the meantime.

sstuercke, 2024-11-14:
It would be good if I could test it.

adlerre, 2024-11-20:
Would you test this branch with your car? If you need help compiling, let me know.

sstuercke, 2024-11-22:
Unfortunately I won't be able to test it until the weekend.

sstuercke, 2024-11-22:
I get an error message during the build.
[build.txt](https://github.com/user-attachments/files/17875749/build.txt)

adlerre, 2024-11-22:
[eUP-Test.zip](https://github.com/user-attachments/files/17876124/eUP-Test.zip) use OTA to update

sstuercke, 2024-11-24:
Hi René,
I tried the firmware and the SOC is coming through in MQTT.
There is only a difference between the car display 64% and MQTT 62%.
But that is because the car display 0% is not battery 0% but 10%.
When will this be included in the original version?

Thanks

adlerre, 2024-11-25:
Hi,
thanks for testing. Sounds great that it works.

I'm planning a complete overhaul of how OBD PIDs are handled. It should be possible to define your own PIDs and save them as a profile. For this I need to implement an expression parser and a GUI to change/add the profile.
I hope to get this done this year.

René

sstuercke, 2024-11-25:
Then I have another suggestion for improvement: make the inputs/outputs controllable via MQTT. I only need the SOC value when the car is connected to the wallbox. I want to switch off the OBD adapter.

closed 2024-12-12 by adlerre