adlerre/obd2-mqtt #26: hardware/assembly guidance

aleks-mariusz, 2024-11-13:
So i saw you have 3d-model STL files, so i got to printing one of them.  It would be really swell if you had any guidance on which files do what (other than the obvious top/bottom ones).  I see there are potentially three different "styles".  Two for the a76xx and one for the sim800.  I assume since i received a GPS module w/ my lilygo order i should use the GPS one, but I'm drawing a blank on the GPS holder file, how it's to be used.  I placed it chip-side down (with the shiney white/pink side face up) and it fits nicely, but falls down into the case when turned upside down so not sure how the gps holder is supposed to keep it in place?

also what screw-hardware are suggested for these size holes and how are they fitted?  

also for example, there's another with 2-sma holes, when would you want to use that ? my gps chip and lte antenna have IPEX, when/why would i want to use a sma adapter/sma-based instead? it's just listed as 'optional' on the main readme without much more detail about what exactly it achieves/advantages of having it? apologies if these are very trivial questions

maybe a bit of a blurb and perhaps an simple assembly drawing or an already-assembled photo in that 3d-files folder would be useful please

adlerre, 2024-11-13:
Hi,

i'm short in time for now to write more. 
Please Download FreeCAD and open the *.FCStd file.

closed 2024-11-14 by adlerre

aleks-mariusz, 2024-11-14:
hey just wanted to say again thanks.. ~do you have any donation links or any buy-me a coffee?~ found it! I really appreciate how quickly you respond and how incredibly fast you put together solutions.  it's an incredibly admirable occurrence!

FWIW, i did download freecad yesterday after your suggestion, but i'm fairly new to cad (i'm on a mac and i've only dabbled a bit in solidworks and now fusion-360 since i don't want to emulate windows any more).. freecad interface looks definitely "open-source" (e.g. a bit confusing) and it took me 20 minutes to figure out how to even make the hidden section visible and orbit around it.. so seeing your new html with the models very simple to view is wayyyy a nicer experience.  just a minor idea (no need to rush to implement) but in the future can you consider using step files? as i think they may be cross-platform/work across-programs, and don't pin the user of a 3d-file to any particular one program please.

adlerre, 2024-11-14:
Thanks!
I also add step files for each case.

I can understand your problem with FreeCAD, I had the same.
But it's worth getting to grips with it!