adlerre/obd2-mqtt #32: No gsm ESP32

Skons, 2024-11-21:
Is it possible to flash an esp32 that does not have a GSM module?

adlerre, 2024-11-21:
No, isn't possible, because the GSM module is required to send data to the Broker.

sskerry, 2024-12-08:
> No, isn't possible, because the GSM module is required to send data to the Broker.

Same question, I have a home assistant running in my camper van, I want to use this locally so no sim card needed. I Just want to connect the ESP to vans wifi and send data directly to the broker running in the van.

adlerre, 2024-12-09:
> First off, amazing project thank you!

Thanks, I appreciate that!

> Same question, I have a home assistant running in my camper van, I want to use this locally so no sim card needed. I Just want to connect the ESP to vans wifi and send data directly to the broker running in the van. I loaded it up on a non-cellular module, the only option missing is the ability to connect to a network, otherwise i think it will work.

There are a few technical reasons for this. An ESP32 shares one antenna for BT and WiFi, which means it can only modulate/demodulate one wireless protocol at a time.
In Espressif, RF coexistence is supported, but this is a bit unstable if you use classic BT.

closed 2025-01-03 by adlerre

thefunkygibbon, 2025-04-17:
I understand that WiFi and BT share the same and it isn't 100% stable but to read sporadically the obd2 values and send them via WiFi isn't going to be a huge issue.  ie using esphome as a Bluetooth proxy works pretty much flawlessly on all of my devices.  sure it might drop a packet here or there, but it's not exactly needing a stable connection to read a bunch of values

thefunkygibbon, 2025-04-17:
ah the ol' "do it yourself" reply.   don't worry, if I had any clue about development I certainly would. but alas my skills lie elsewhere and one doesn't have the time and patience to learn whole new skills like this at my age.    

thanks anyway and good luck with your project.