adlerre/obd2-mqtt #34: failed to obtain VIN

aleks-mariusz, 2024-12-01:
I got my obd2-mqtt code loaded, got my sim card activated and popped in, i got the gsm and gps antenna's connected, plugged in the bluetooth and... 

failed to obtain VIN
failed to obtain VIN
failed to obtain VIN

repeatedly... 

attached is the full log.. [esp-web-tools-logs (10).txt](https://github.com/user-attachments/files/17969795/esp-web-tools-logs.10.txt)

anyway, i can't tell if this is something that is causing it to not continue/hang indefinitely, or if it is continuing anyway and there's something weird about the serial console (when i request to reset the device, i see a efw more messages that didn't appear until the split-second i clicked reset, so maybe some serial-output-buffering or something?).. the bluetooth device (obd2 scanner) blinks continuously tho, so i presume data is being sent back and forth.. 

i don't see anything in my mqtt server tho (unless i'm not looking in the right place).

adlerre, 2024-12-02:
Sorry for the late reply. 
This VIN lines can be ignored. You're right, the problem was the connection to the MQTT broker.

aleks-mariusz, 2024-12-16:
Hi @adlerre i had a chance to update my lilygo w/ the new firmware that has support for ssl.

Alas, I'm still having issues connecting.  I see this repeatedly:

```
The client obd2mqtt-0123456789ba connects to the MQTT broker......failed with state -2
```

I also tried disabling SSL (and changing port to 80), that seems to give a different error:

```
The client obd2mqtt-0123456789ba connects to the MQTT broker......failed with state -4
```

Any advice what these failure codes mean? What can be tried? or if you have a debugging build you want me to give a go?

adlerre, 2024-12-16:
Hi @aleks-mariusz,

Error -2 means "MQTT_CONNECT_FAILED - the network connection failed" and -4 means "MQTT_CONNECTION_TIMEOUT - the server did not respond within the keepalive time".

You have several options to check the connection:

Use nmap to check the ports:

nmap <hostname/ip> -p 80 (http) / nmap <hostname/ip> -p 443 (https) or use any other port after -p.

Check the Mosquitto Broker logs, they should give you a clue as to what is wrong.

Try connecting with [MQTT Explorer](https://mqtt-explorer.com).

If that works, we need to dig deeper.

closed 2025-02-03 by adlerre

aleks-mariusz, 2025-02-03:
i saw you had updated the VIN-checking code in 0.14 so thanks for that.  I still have my connectivity issues but i'll open a new case about that as that convo is getting quite a bit off-topic

adlerre, 2025-02-03:
That reading of the VIN wasn't necessary/not working on the most cars, so i decided to remove this.
It's okay to create a new case for that. So we can help others with the same issue.