adlerre/obd2-mqtt #35: Is any of the builds compatible with Lilygo T-A7670E?

spitfire, 2024-12-06:
I got a Lilygo T-A7670E (https://lilygo.cc/products/t-sim-a7670e) with built in GPS and LTE, can it be used with any of the builds you're providing?

adlerre, 2024-12-06:
Have you tried the A7670X build?

marcoraddatz, 2024-12-13:
Yes, it is. I have the same product.

spitfire, 2024-12-18:
I finally got around to configuring MQTT broker and finally  T-A7670E. I've used the A7670X build of 0.11, but it fails to connect to my ELM 327 and I don't know why. What can I do to troubleshoot it?

I've connected to serial via USB to grab the logs:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (786) esp_core_dump_flash: No core du����ѥѥ���found!
E (787) esp_core_dump_flash: No core dump partition found!
Start Access Point...WiFi AP stopped.
WiFi AP started.
AP - IP address: 192.168.4.1
HTTP server started
Start modem...
Initializing modem...
Modem Name: A7670E-FASE
Modem Info: Manufacturer: SIMCOM INCORPORATED Model: A7670E-FASE Revision: A7670M7_V1.11.1 IMEI: 862771075502610
Waiting for network......success
Network connected
Connecting to internet......success
GPRS connected
IP Address: 100.72.48.250
Enabling GPS/GNSS/GLONASS......success
Discover Bluetooth devices...
>>>>>>>>>>>Found a new device: Name: OBDII, Address: 00:10:cc:4f:36:03, cod: 0x10091c, rssi: -48
Stopping discover...stopped
Search device: OBDII
 ----- 00:10:cc:4f:36:03  OBDII -48
scanned for services, found 1
     channel 1 (JL_SPP)
connecting to 00:10:cc:4f:36:03 - 1
Couldn't connect to OBD scanner - Phase 2
```

The lights on it stop blinking and turn solid (as if the connection was established) when  ESP32 is trying to connect, but no dice.
This is the adapter I'm using:
[image]

spitfire, 2024-12-18:
> Yes, it is. I have the same product.

thanks! A bit off topic, but do you have a case for it? I have a 3d printer, but don't know how to create any models really.

spitfire, 2024-12-18:
> Yes, it is. I have the same product.

@marcoraddatz  thanks! A bit off topic, but do you have a case for it? I have a 3d printer, but don't know how to create any models really.

adlerre, 2024-12-19:
> The lights on it stop blinking and turn solid (as if the connection was established) when ESP32 is trying to connect, but no dice. 

This is a problem with this ELM327 adapter, it is too slow. Phase 2 means it is connecting to the car. If the lights are solid, it is trying to get the right protocol. You have 2 options: be patient or go to settings and select the right connection protocol.
If you want to avoid these problems, use a better adapter. I stopped using this one in a while because of these problems.

I am currently using this one. [Adapter](https://www.amazon.de/gp/product/B0BG4W6MLD)

> @marcoraddatz thanks! A bit off topic, but do you have a case for it? I have a 3d printer, but don't know how to create any models really.

You can try this -> [case](https://github.com/adlerre/obd2-mqtt/tree/main/3d-files#for-a7670-boards-with-battery-and-external-gps4g-antenna)
But I'm not done yet, it's for my new device which I want support.

spitfire, 2024-12-19:
> This is a problem with this ELM327 adapter, it is too slow. Phase 2 means it is connecting to the car. If the lights are solid, it is trying to get the right protocol. You have 2 options: be patient or go to settings and select the right connection protocol. If you want to avoid these problems, use a better adapter. I stopped using this one since a while because of these problems.

By connection protocol you mean the one adapter talks to my car via OBD and is dependent on the model of my car? Any idea what would I use for an Opel Astra H?

> I am currently using this one. [Adapter](https://www.amazon.de/gp/product/B0BG4W6MLD)

Ooo, that is really pricey (about 50x more than this one) and it would more than double the cost of the whole project (ESP32 was ~32$) and make it really not viable to me. It is also huge, and would not fit and let me close the lid over the adapter. If setting the protocol manually doesn't work are there any other options you would recommend that would place me between 1,29$ and 49€? ;)

I have one more adapter that works with my iPhone, but wasn't getting discovered by ODB2MQTT, I guess it won't work with it?
[image]

> You can try this [case](https://github.com/adlerre/obd2-mqtt/tree/main/3d-files#for-a7670-boards-with-battery-and-external-gps4g-antenna). But I'm not done yet, it's for my new device which I want support.
Thanks, I'm printing it now and will give it a try! I found https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/dimensions/T-A7670X-Shell.7z on manufacturer's GitHub but it's terribly overcomplicated (a lot of knobs and small parts for no good reason) and not practical - it seems to have a place to but the GPS antenna in...but it doesn't fully fit (and its cable is too short to reach out through the hole to be placed outside).

adlerre, 2024-12-19:
> By connection protocol you mean the one adapter talks to my car via OBD and is dependent on the model of my car? Any idea what would I use for an Opel Astra H?

Protocol ISO_15765_11_BIT_500_KBAUD should work!

> Ooo, that is really pricey (about 50x more than this one) and it would more than double the cost of the whole project (ESP32 was ~32$) and make it really not viable to me. It is also huge, and would [...]

You can try [this](https://www.amazon.de/gp/product/B0D477PZNH), but I don't know if it works.
 
> I have one more adapter that works with my iPhone, but wasn't getting discovered by ODB2MQTT, I guess it won't work with it?

Is there a button to enable pairing? If so, press and let it detect, or you can get the MAC address of the device from your iPhone.

spitfire, 2024-12-19:
> Protocol ISO_15765_11_BIT_500_KBAUD should work!

Thanks, I'll try it!

> Is there a button to enable pairing? If so, press and let it detect, or you can get the MAC address of the device from your iPhone.

There isn't one, I will try to work with it if the blue one doesn't work - maybe it was pairing to my phone and not being discoverable because of that.

An update on your case - some of the holes don't fit - I've aligned it to the screw holed at the end (bottom ones on this photo) of the box:
[image]

Switch cutout is fine:
[image]

Port openings are off (especially one for USB-C):
[image]
The ports are too far from the openings for the plug to go into the socket (at least for microUSB):
[image]

Do you think I could hot glue the GPS antenna to the bottom of the case? Its cable doesn't even let it reach the side of the case where the cable openings are:
[image]

adlerre, 2024-12-19:
I'm using this with a external LTE/GPS antenna.

[image]

In the next few days I'll build a new one with better connection holes and support for an internal GPS antenna.

adlerre, 2024-12-25:
A little Christmas present for you!  [Case](https://github.com/adlerre/obd2-mqtt/tree/main/3d-files#for-a7670-boards-with-battery-and-internal-gps4g-antenna)

Merry Christmas!

[image]

spitfire, 2024-12-25:
> A little Christmas present for you! [Case](https://github.com/adlerre/obd2-mqtt/tree/main/3d-files#for-a7670-boards-with-battery-and-internal-gps4g-antenna)
> 
> Merry Christmas!
> 
> ![case](https://private-user-images.githubusercontent.com/1357498/398593528-8d4d3eca-ab80-4689-9f87-42bb24611bb8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUxNTg4MTIsIm5iZiI6MTczNTE1ODUxMiwicGF0aCI6Ii8xMzU3NDk4LzM5ODU5MzUyOC04ZDRkM2VjYS1hYjgwLTQ2ODktOWY4Ny00MmJiMjQ2MTFiYjguanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjVUMjAyODMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjNjODEyZTZmNGRjY2I5YjM1YTM5Y2JlZjU1MzBhNDkxOWUxYzg4NGRlNGUyNDdlMWRlOTQwYjZlMGE3MDQyMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.KWRSKfx_OYSatVpuztmXMtWvDPAYyx0ksWCTnlKEt-c)

That looks great, Merry Christmas to you too!

closed 2025-01-10 by adlerre