adlerre/obd2-mqtt #45: Using Helium Lora - Tbeam

old-square-eyes, 2025-01-10:
Hi. I am getting my head around this solution - it looks awesome. I have an ELM and a Nissan Leaf. I want to get data back to my Home Assistant via Helium Lora.

I already have a Helium GPS tracker based on the TBeam. Here is a reference implementation...

https://github.com/Max-Plastix/tbeam-helium-mapper/

The LilyGo TTGO T-Beam v1.1 has Bluetooth.

How can I get this device supported, to read my ELM and send OBD data back to Home Assistant via Lora/Helium?

adlerre, 2025-01-11:
Hello. I currently have no idea about LoRa/LoRaWan and need to dig deeper to learn more about it.
Therefore, I cannot currently say whether I can/will support devices with LoRa.

adlerre, 2025-01-12:
After reading some documents about LoRaWAN/Helium and LoRa in general, I can say that I don't want to support this.
The main goal of this project is to send telemetry data while driving and preferably live.
So I see two main problems with using LoRa for this. The first is global coverage, it's not nearly as good as GSM/GPRS/LTE.
The second is the amount of data that can be sent. In an hour, the main profile (OBD) sends an estimated 1.5MB of data over the cellular network. This doesn't seem to be possible with LoRa due to regulations regarding airtime in some countries.

closed 2025-01-12 by adlerre