adlerre/obd2-mqtt #47: Using WIFI as hotspot

Rick45, 2025-01-23:
Hi,
First of all, really nice work with the project, i received the device yesterday and I'm implementing this one Renault Zoe.

I'm fairly new to the ESP32 implementations, so sory if this is something that is not possible with this type of devices.

I noticed that when i connect the phone to it there is a message saying that some task have been put on hold.
Is possible to use the wifi module as an hotspot? 

My use case is the following.

I live in a apartment where my wifi does not reach my parking spot, so i was thinking if is possible to use the wifi to share internet to a Shelly 1PM so i can read and interact with it from my main home assistant instance.

Thanks.

adlerre, 2025-01-23:
Hi,

thanks, I appreciate that.

If I understand you correctly, you want to send the data via WiFi instead of the cellular network. That doesn't really work because the ESP32 shares the antenna for BT and WiFi (see #32).

This solution requires a connection to a cellular network to send OBD data to the MQTT server while driving.

Regards
René

Rick45, 2025-01-23:
Not directly. The idea was to allow devices connected to the WiFi module to access the Internet provided by the GSM module.
Basically using the WiFi module as an Internet Hotspot.

adlerre, 2025-01-24:
Ok, I understand. But here same problem, either OBD data can be read or the AP can be used.
When the AP is used, reading the readings will stop to avoid BT connection problems.
On the other hand, there is not much space left in the flash to implement this feature.

adlerre, 2025-01-24:
oh, thank you, that would be nice!