adlerre/obd2-mqtt #48: Waveshare ESP32-S3 A7670E

abonforti, 2025-01-23:
Hi,
quick question: found this amazing project and would like to implement the same in my car. Do you happen to know if the combination of [Waveshare ESP32-S3 A7670E](https://www.amazon.it/Waveshare-Development-Bluetooth-Telephone-Positioning/dp/B0CVVY38JG/) and [VGate iCar PRO](https://www.amazon.it/gp/product/B07SQLY39V/) is actually supported by this firmware?

Thanks in advance!

adlerre, 2025-01-24:
Hi,

thanks. Have you tried using the T-A7670X build? The modem should be the same. The BT dongle should work.

I'm trying to get one of those boards and see if it's possible to support this.

abonforti, 2025-01-24:
Hi, not really. I just found this project and started searching for Amazon links which can be compatible with your build, that's the reason of the ask.

adlerre, 2025-01-25:
Okay, after some research I can confirm that this device would not work with this project. The ESP32-S3 only supports BT LE, but BT Classic is required for that.

closed 2025-01-25 by adlerre