adlerre/obd2-mqtt #50: [Q] How to fill  / read ODB fields

Rick45, 2025-02-13:
Hi, 
i'm trying to understand how all of this ODB2 fields work.
I was able to have this working and using the VW profile i'm able to read the Battery Voltage.
 With the help of the CanZe repo i was  able to find the "fields"  for my Zoe, but as i dont understand much of this so i'm struggle a bit of how to fill them in the config.

for the  example bellow we have for the ID 42e  (1070 SID?)  we have the corresponding bit intervals that represent the value what we want to read.

[image]

this  maps  i any way to the  Service/ SID and for the Scale Factor/ Bias

what is the "best" way to test / find  the correct codes?
Currently i  don't have my  laptop so I'm mostly moving from the Desktop to the Car to test changes :P

thanks :)

adlerre, 2025-02-16:
Hi,

I'm doing some searching to find the right PIDs for your Zoe and create a profile. But I can't test it.

At the moment I don't know what these numbers in your table mean. I need to look at the source code of CanZe.

Can you please test this profile?

[states.json](https://github.com/user-attachments/files/18814433/states.json)

Rick45, 2025-02-16:
Thanks, i will test it today. I have a zoe r90, a phase 1. how did you end with this values? 
the table that i have attached was from the CanZe fields list: 
[https://github.com/fesch/CanZE/blob/master/app/src/main/assets/ZOE/_Fields.csv](https://github.com/fesch/CanZE/blob/master/app/src/main/assets/ZOE/_Fields.csv)

i'm going to the canze code to try to see how can i convert them to here .

adlerre, 2025-02-16:
Er, if I'm honest, from the competition. ;)
https://github.com/meatpiHQ/wican-fw/tree/main/vehicle_profiles

Okay, i build another profile, but now for PH1.

[states-zoe-ph1.json](https://github.com/user-attachments/files/18815038/states-zoe-ph1.json)

closed 2025-04-11 by github-actions[bot]