adlerre/obd2-mqtt #51: ESP keeps sending same data over again

MorganMLGman, 2025-02-16:
Hi,
I have a problem that ESP keeps sending same data to MQTT over and over again. Seems like new data read from OBD is sent only once and after that ESP keeps sending same data.

These graphs should explain better what I am talking about.
[image]

Engine RPM is always 821 end I was driving around parking lot. New data was sent only when I restarted ESP.

My ESP32 is: T-Call-A7670X-V1-0
OBD: [iCar2 BLE 4.0 OBD2](https://www.amazon.com/vLinker-Bluetooth-Adapter-Diagnostic-Black/dp/B0D474RC4W?crid=24QNB85R5RZ83&dib=eyJ2IjoiMSJ9.XzBErKSp-QQKaOOmoMRHCtY9Hxll8r7h9prKAMk6bx4MIrRQqYZsqqC-pqMn-xNdHyQ56nmBPrNIrQuX5NNBAx9m6T81V3qmFdXXqKxBgBQlLlb6EiJmYhZR7FXwQrwF_Am9JNqZcEFTd3krZgMS70ZCy6Jrea3xc_BBMvq_EmPrk20fEmapbfcw1s2Xy9GzJ8I6c0Z7O-KMtbWMPsySnIDlMGNu-Kl-JjXrpRJWk5dRpCwxTebpIhZibVLLugZ8cuLbgDAc8cWyNIV9Nl9lVhjJMAg0hUiFwl6NZofGTxB5B88LowrN_cGor9u-2JEmB9RZT-Q0Xk1BL7uIVSY_ObU9ZvYnFzCiprH347rIWsMZSlRafBPnq5JRHL2LbtEKYrqIftij28yg_xETVLF3aAjQHcw2x98MwsZTMS-a-1rmnxQ91Z62YUmzyjdu42_x.RKW9kY35HusOR-84vwzFnRq44T7gmlVh6l8QnuAzFWE&dib_tag=se&keywords=vgate%2Bobd2&qid=1739744431&sprefix=vgate%2Bobd2%2Caps%2C208&sr=8-48&th=1)
Car: Honda Accord VII

I tested OBD with app "Car Scanner" and it shows correct data in real time

MorganMLGman, 2025-02-16:
Also how to get this data from MQTT to HA? 

[image]

Do I need to manually create sensors from MQTT topics in configuration.yaml file in HA or is there some template or ready integration?

aw1604, 2025-02-17:
> Also how to get this data from MQTT to HA?
> 
> Do I need to manually create sensors from MQTT topics in configuration.yaml file in HA or is there some template or ready integration?

Hello,

In HA go to Integrations - MQTT - Devices, then search for a device by "Rene Adler" or search for the MAC of your ESP32 device. Click on that device and you should see Sensor / Diagnose data ready to add to a Dashboard.

HTH,
Andreas

adlerre, 2025-02-17:
Hi,

have you any console logs or some output from the MQTT integration? I would want the output from MQTT Info link, this shows incomming data as YAML/JSON. The most interesting part is the "*_connection" payload.

René

MorganMLGman, 2025-02-17:
This is last 10 message

[image]

I have set data send interval to be 3s

adlerre, 2025-02-17:
That seems to be fine. Are there any differences in *_rpm topic?

adlerre, 2025-02-17:
These values ​​are too high. What protocol have you set?

MorganMLGman, 2025-02-17:
Auto

MorganMLGman, 2025-02-17:
In "Car Scanner" app I am using this "default" profile and it seems to work

[image]

adlerre, 2025-02-17:
Can you please try it with protocol ISO_15765_11_BIT_500_KBAUD.

MorganMLGman, 2025-02-17:
Ok, will try today.

MorganMLGman, 2025-02-17:
I tried ISO_15765_11_BIT_500_KBAUD, but it is not working at all, only 12 diagnostic topics from esp are sent to mqtt. Data from OBD like rpm, engine_running is unavailable. 

With AUTO protocol correct data is sent only once.

[image]

1340 is correct engine rpm and then it goes to 638500865

MorganMLGman, 2025-02-17:
[image]

[image]

adlerre, 2025-02-21:
I bought an iCar2 BLE 4.0 OBD2 dongle and tested it to see if it has any issues. But it works fine.
So we need more information to fix your problem. Can you send me a console log? What values ​​do the "*_supportedPids_*" topics contain?

MorganMLGman, 2025-02-21:
I'm on vacation right now. Will update on Monday.

MorganMLGman, 2025-02-24:
[image]

[image]

Protocol AUTO

Engine rpm was 821 then jumped up.

[image]

[image]

adlerre, 2025-02-26:
Okay, the logs shows no issues. The topic supportedPids_1_20 shows that RPM should supported. But to really confirm that, the check "check if PID is supported" within settings tab.

Any other car related topics shows valid values?

MorganMLGman, 2025-02-27:
I see no difference with or without "check if PID is supported"

Protocol: Auto, check if PID is supported ✅  ##
[image]

Protocol: Auto, check if PID is supported ❌ 
[image]

Protocol: ISO_15765_29_BIT_500_KBAUD check if PID is supported ✅
[image]

Protocol: ISO_15765_29_BIT_500_KBAUD check if PID is supported ❌ 
[image]

Only these 2 protocols seems to be "working", I tried all and rest returns "Send OBD data...fail" on serial.

> Any other car related topics shows valid values?
Battery voltage has correct values

RPM, engine coolant temperature, air intake temperature either have correct value once and then they don't refresh or have way to high values.

adlerre, 2025-03-02:
Sorry for the late reply.
I have attached a debug build for you that prints the data to the console. To get information on what is wrong, disable all OBD states except RPM, coolant temperature, and air intake temperature.
The console output will show the state name and the previous and current value.

[firmware-A7670X-V1-0.zip](https://github.com/user-attachments/files/19042394/firmware-A7670X-V1-0.zip)

MorganMLGman, 2025-03-02:
No problem, I am glad for your help.

I have logs, there are 4 sections with esp reboot between them.

1. Auto
2. ISO_15765_29_BIT_500_KBAUD
3. ISO_15765_11_BIT_500_KBAUD
4. Back to auto

[image]

MorganMLGman, 2025-03-02:
MQTT also seems to have correct data with AUTO now.

[image]

adlerre, 2025-03-03:
Okay, looks good. Now you can enable step by step other states and on last the calculated ones.

MorganMLGman, 2025-03-03:
Ok so I think I found which measurements cause issues. 

I disabled: 
- oilTemp
- throttle
- relativePedalPos
- ambientAirTemp
- fuelRate
- fuelType
- manifoldPressure
- timingAdvance
- odometer (not sure if this needs to be disabled)

After that I was able to get it to work. It was working for almost 5 mins without noticeable issues.
[states.json](https://github.com/user-attachments/files/19060167/states.json)
[serial_20250303_192806.txt](https://github.com/user-attachments/files/19060179/serial_20250303_192806.txt)

[image]

adlerre, 2025-03-20:
Any news about this? Have you tested some calculations?

MorganMLGman, 2025-03-20:
Since my last comment it is working without any problems so far, I flashed 0.19.2, but with disabled calculations and sensors mentioned above as they are not available. I need to once again borrow obd from @cichy1173, if I remember correctly his obd was reading more sensors from car.

MorganMLGman, 2025-03-24:
Can't get any more readings, tried with different obd Vgate iCar Pro. I think it is all I can get from this car. Using other obd "car scanner" app suggested that I should be using protocol ISO-9141-2, tried that in the app and more sensors showed up, but these new sensors never filled with data, they stayed blank or always 0. 

@adlerre Do you think it would be possible and more less accurate to use mass air flow sensor to calculate fuel usage? I think that air to fuel ratio should be rather constant and google says that it should be around 14.5:1 in diesel engine. I could use that and create custom sensor to calculate fuel g/s and then fuel L/s using fuel density that also should be around 820-845kg/m3 and then L/100km using speed sensor. It should be possible, but I don't know if it would be any accurate.

adlerre, 2025-03-25:
I see the MAF values in your scan, so it's easy to use the current implemented calculation. You only need to set the fuel type. Like so:

```$consumption + ($mafRate * 3600 / (afRatio($fuelType) * density($fuelType))) / 3600 * ($millis - $consumption.lu) / 1000```

```$consumption + ($mafRate * 3600 / (afRatio(4) * density(4))) / 3600 * ($millis - $consumption.lu) / 1000```

The 4 stands for diesel. [See](https://github.com/adlerre/obd2-mqtt/blob/cc1a06ce72a70373914661200dec61e6dabfb542/src/obd.h#L57)

MorganMLGman, 2025-03-27:
Thank you for your help, I would not figure it out without help. 

I enabled consumption and added fuel type, and it is working, but calculated values are too high.
I drove around 110km on highway and consumption calculated by car is 5.9L/100km, which is more/less accurate with my previous drives and calculations I make between refueling. Value calculated by esp is 2x higher. But I don't have fuel rate sensor nor actual air/fuel ratio sensor, so I guess I don't have much choice, but to multiply whole consumption formula by some value like 0.5, and with trial and error get close to real values.

[image]
[image]

Thank you for help and I think this issue could be closed.

adlerre, 2025-03-28:
you're welcome

closed 2025-03-28 by adlerre