adlerre/obd2-mqtt #52: Diag data received in MQTT, but HA showing unavailable

aw1604, 2025-02-17:
Hello,

First of all : this is a fantastic project, thank you very much for your work !

My setup:
- SIM800L running firmware 0.19.0 (ordered a A7670, but stuck in transit.. and had the 800L available from another project)
- ELM327 OBD
- Car: Skoda Octavia 5E 1.6 TDI (so interested in some DPF / ash-load values in the future..)
- mosquitto with 2 listeners (1883 internal, 8883 external - planned for TLS, but using username/password for now). Tried "old reliable" mosquitto-2.0.11-openssl and 2.0.20-openssl
- HA 2025.2.4

Issue:
Upon connect, I get the Diagnostic data shown fine for a minute in HA. After this, all Diag/Sensors are shown as "unavailable" in HA.

[image]

In the mosquitto logs I can see the Diag data still incoming every 30 seconds, and published to HA:

[image]

Any hints on this issue ? I had suspected mosquitto not handling multiple listeners correctly, thats
why I tried different versions of mosquitto.

Best Regards,
Andreas

adlerre, 2025-02-17:
Hi Andreas,

is a little bit strange. If you have incoming data in MQTT this should also displayed in HA.
Data marked as unavailable if the "*_connection" payload contains off or last connection time expired. Can you take a look on "MQTT Info" link (within MQTT integration).

René

aw1604, 2025-02-17:
Thanks ! the last *_connection messages show all  Payload: connected

[image]

adlerre, 2025-02-17:
Can you see any changes in the messages?

[image]

aw1604, 2025-02-19:
Well, after doing more tests it seems the ESP32-SIM800L board is defective. It sends data for a few minutes when its cold and fresh started, but then just ceases, and way worse when its hot.

I'll dump this board and wait for my ordered A7670E, and then try it again..

Best Regards,
Andreas

adlerre, 2025-02-21:
Thanks for this info. I hope you have better luck with the next device.