adlerre/obd2-mqtt #53: How to pair ELM327 with password?

naamah75, 2025-02-23:
How do I pair an ELM327 bluetooth adapter that asks for a PIN?

adlerre, 2025-02-23:
This is not currently supported. However, it is possible to implement it.
Which dongle should be used? (to get one)

naamah75, 2025-02-23:
I don't remember the model well, but it is a common ELM327 of those of which you can find plenty on Amazon. I did a short search, again on Amazom and it seems that they all require a PIN like 1234 or 000

naamah75, 2025-02-23:
Do you think that such a modification, which is very crude, would be fine
`    if (!serialBt.begin("OBD2MQTT", true)) {
        Serial.println("========== serialBT failed!");
        ESP.restart();
    }
    serialBt.setPin("1234");`

adlerre, 2025-02-23:
> I don't remember the model well, but it is a common ELM327 of those of which you can find plenty on Amazon. I did a short search, again on Amazon and it seems that they all require a PIN like 1234 or 0000

I have tested some, but none required a pin.

> Do you think that such a modification, which is very crude, would be fine
> 
> ` if (!serialBt.begin("OBD2MQTT", true)) { Serial.println("========== serialBT failed!"); ESP.restart(); } serialBt.setPin("1234");`
> 
> ... into the obd.cpp

Yes, this would work. But I would add this to settings page and set if not empty.

naamah75, 2025-02-23:
Perfect! So could you recommend some model of adapter that you have tried and does not require pins? in any case as soon as I have a way, if you want, I will write you the model of my ELM327 adapter

Yes, implementing the configuration on the home screen looks good and that's what I was trying to do, but it's your code! And surely you will be better, it takes me much longer to understand where to put my hands

adlerre, 2025-02-23:
> Perfect! So could you recommend some model of adapter that you have tried and does not require pins? in any case as soon as I have a way, if you want, I will write you the model of my ELM327 adapter

https://www.amazon.de/dp/B0BG4W6MLD
https://www.amazon.de/dp/B0D474RC4W

naamah75, 2025-02-23:
Thank you for your patience, you were very kind. Do you think you can change the PIN in the settings in the near future?

naamah75, 2025-02-23:
[image]

naamah75, 2025-02-24:
the modification i made doesn't work, the device is detected with its mac address, but with name "0" instead of "OBDII"... i don't know if this can help. Anyway the bluetooth connection doesn't happen. If i can help you with the tests let me know!

> I have tested some, but none required a pin.
>
>
> Yes, this would work. But I would add this to settings page and set if not empty.

adlerre, 2025-02-24:
Have you tested with PR #54? Can you provide some logs?

adlerre, 2025-02-26:
> For this model the manufacturer states that you must enter the PIN: [Vgate iCar Pro Bluetooth 4.0 (BLE) ](https://www.amazon.it/dp/B071D8SYXN/?coliid=I380ODGJV37TZJ&colid=14U1O7H2Y5OQ&ref_=list_c_wl_lv_ov_lig_dp_it&th=1)

I haven't tested, but I think it's not required.

> 
> For the model you suggested, which does not use a PIN, however, I read that you have to press the power button every time and it does not start automatically with the car; is that so? [Vgate iCar2 BLE 4.0](https://www.amazon.it/dp/B0D474RC4W/?coliid=I1M6Q8FAQGH1I4&colid=14U1O7H2Y5OQ&ref_=list_c_wl_lv_ov_lig_dp_it&th=1)

I used this this device without any issues, no pin required and it's not required to press the power button on any use. I think you must power on, only on low battery.

> Googling a bit I noticed conflicting opinions of users who said that for the same, for some users the PIN was required and for other users not; I discovered that the discriminating thing for many is the operating system of the smartphone, it seems that the pairing for Apple is very often without PIN (I use android)

https://www.amazon.de/dp/B0BG4W6MLD using this since months, without any issues. No pin, no button press is needed. The button is only required to pair the dongle. Switches dongle in visible mode.

adlerre, 2025-02-26:
> Can I help you with some more attempts?

Not really, because you did everything I would have done.

naamah75, 2025-03-16:
me again :)
I ordered this OBD adapter
[https://it.aliexpress.com/item/1005005704073093.html?spm=a2g0o.order_list.order_list_main.23.6e7b3696z8c38w&gatewayAdapt=glo2ita](url)
and it actually doesn't require the pin, but I think there's still some problem...

there are some changes in the log compared to before but still unable to establish the connection :(
This is the log:

```
Modem Name: A7670E-FASE
Modem Info: Manufacturer: SIMCOM INCORPORATED Model: A7670E-FASE Revision: A7670M7_V1.11.1 IMEI: 862771076817991
Waiting for network......success

Network connected
Connecting to TM......success
GPRS connected
IP Address: 10.59.46.50
Enabling GPS/GNSS/GLONASS......success
scanned for services, found 1
     channel 1 (Port)
connecting to 99:99:1b:35:05:89 - 1
Bluetooth disconnected.
Couldn't connect to OBD scanner - Phase 1
Restarting OBD connect.
scanned for services, found 1
     channel 1 (Port)
connecting to 99:99:1b:35:05:89 - 1
Couldn't connect to OBD scanner - Phase 1
Bluetooth disconnected.
Restarting OBD connect.
scanned for services, found 1
     channel 1 (Port)
connecting to 99:99:1b:35:05:89 - 1
Bluetooth disconnected.Couldn't connect to OBD scanner - Phase 1

Restarting OBD connect.
scanned for services, found 1
     channel 1 (Port)
connecting to 99:99:1b:35:05:89 - 1
Couldn't connect to OBD scanner - Phase 1
Bluetooth disconnected.
Disconnected-- 
```

so i downloaded an app called ELM327 identifier to figure out what protocols it is using and this is the log:

```
ELM327 Identifier 1.18.19
Data scansione=2025-03-16 05:31:47
Nome dispositivo=KONNWEI
Versione dispositivo (dichiarata)=ELM327 v1.5

Comando	Versione	Risultato	RAW
AT@1	1.0	true	AT@1\rOBDII to RS232 Interpreter\r\r>	
ATSPA6	1.0	true	ATSPA6\rOK\r\r>	
ATTPA6	1.0	true	ATTPA6\rOK\r\r>	
ATTP6	1.0	true	ATTP6\rOK\r\r>	
ATSP00	1.3	true	ATSP00\rOK\r\r>	
ATSP6	1.0	true	ATSP6\rOK\r\r>	
ATAL	1.0	true	ATAL\rOK\r\r>	
ATNL	1.0	true	ATNL\rOK\r\r>	
ATAMC	2.0	false	ATAMC\r?\r\r>	
ATAMT20	2.0	false	ATAMT20\r?\r\r>	
ATRA70	1.3	true	ATRA70\rOK\r\r>	
ATSR70	1.2	true	ATSR70\rOK\r\r>	
ATAR	1.2	true	ATAR\rOK\r\r>	
ATAT0	1.2	true	ATAT0\rOK\r\r>	
ATAT2	1.2	true	ATAT2\rOK\r\r>	
ATAT1	1.2	true	ATAT1\rOK\r\r>	
ATBD	1.0	true	ATBD\rFD 80 FF 11 7F 00 FF 00 01 7F 82 EF 0C \r\r>	
[... 85 lines cut ...]
	20:FF F  21:FF F  22:FF F  23:FF F
	24:00 F  25:00 F  26:00 F  27:FF F
	28:FF F  29:FF F  2A:38 F  2B:02 F
	2C:E0 F  2D:04 F  2E:80 F  2F:0A F
ATR0	1.0	true	ATR0\r\nOK\r\n\r\n>	
ATR1	1.0	true	ATR1\r\nOK\r\n\r\n>	
ATRV	1.0	true	ATRV\r\n14.0V\r\n\r\n>	
ATS0	1.3	true	ATS0\r\nOK\r\n\r\n>	
ATS1	1.3	true	ATS1\r\nOK\r\n\r\n>	
ATSDFF	1.4	true	ATSDFF\r\nOK\r\n\r\n>	
ATRD	1.4	true	ATRD\r\nFF \r\n\r\n>	
ATSH000007E0	2.0	false	ATSH000007E0\r\n?\r\n\r\n>	
ATSHA848F1	1.0	true	ATSHA848F1\r\nOK\r\n\r\n>	
ATSH7B0	1.0	true	ATSH7B0\r\nOK\r\n\r\n>	
ATST32	1.0	true	ATST32\r\nOK\r\n\r\n>	
ATSWFF	1.0	true	ATSWFF\r\nOK\r\n\r\n>	
ATTAF9	1.4	true	ATTAF9\r\nOK\r\n\r\n>	
ATV1	1.3	true	ATV1\r\nOK\r\n\r\n>	
ATV0	1.3	true	ATV0\r\nOK\r\n\r\n>	
ATWMC133F13E	1.0	true	ATWMC133F13E\r\nOK\r\n\r\n>	
ATPC	1.0	true	ATPC\r\nOK\r\n\r\n>	
ATWS	1.0	true	ATWS\r\n\r\rELM327 v1.5\r\r>	
ATD	1.0	true	ATD\r\rOK\r\r>	
ATZ	1.0	true	ATZ\r\r\rELM327 v1.5\r\r>	
```

adlerre, 2025-03-16:
Car engine is on?
Have you tried to set the OBD protocol by hand, in most cases ISO_15765 _11_BIT_500_KBAUD sould work?

naamah75, 2025-03-16:
yes, the engine of the car was running, and in fact I managed to make the log that I posted above and I also made a connection with CarScanner from which you can see the protocol that you suggest, and that I also tried to set in manual.

[image]

adlerre, 2025-03-16:
Can you try this with below firmware.

[firmware.bin-T-A7670X.zip](https://github.com/user-attachments/files/19270120/firmware.bin-T-A7670X.zip)

I hope this is the right device.

naamah75, 2025-03-17:
Yes, the device is correct, but it seems that it doesn't really want to communicate with my adapter.
is there any debugging I can do?

adlerre, 2025-03-17:
Upload (use OTA) this firmware, it has the needed debug output.

naamah75, 2025-03-17:
that's what I did, but there's nothing different from the log I posted above

adlerre, 2025-03-17:
The ESP is in the near of the adapter? Please remove BT adapter MAC from settings and post output from BT scanning.

adlerre, 2025-03-20:
Looks good. To test it, I ordered the adapter from AliExpress.

naamah75, 2025-03-22:
You are very kind, thank you. If you prefer I can also send you the OBD adapter.

adlerre, 2025-03-25:
I received the adapter I ordered, and it's absolute crap! I can't establish a working connection. Discovery works, but the connection doesn't!

naamah75, 2025-03-25:
it doesn't work with any android app? I use CarScanner and ELM327 identifier (on Android)

adlerre, 2025-03-26:
> I received the adapter I ordered, and it's absolute crap! I can't establish a working connection. Discovery works, but the connection doesn't!

The initial frustration disappeared after testing this connector with Car Scanner Pro. It works as expected, but not with BT Classic, it requires a BLE connection. This is currently not supported. I try to get it working, but this takes some time.

adlerre, 2025-04-01:
I have two pieces of news for you: one good and one bad.

The good one...

```
Enabling GPS/GNSS/GLONASS......success
connecting to 99:99:1b:35:03:0f
Connected to ELM327
Read location......done (1505ms)
The client obd2mqtt-99991b35030f connects to the MQTT broker......connected.
Send discovery data......done (3678ms)
Send diagnostic discovery data......done (1315ms)
Send static diagnostic discovery data......done (567ms)
Send diagnostic data......done (226ms)
Send static diagnostic data......done (183ms)
Send location data......done (73ms)
Send OBD data......done (702ms)
Send OBD data......done (716ms)
Send OBD data......done (634ms)
Send OBD data......done (660ms)
```

...and the bad one

```
Retrieving maximum program size .pio/build/T-A7670X/firmware.elf
Checking size .pio/build/T-A7670X/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.3% (used 66516 bytes from 327680 bytes)
Flash: [==========]  99.9% (used 1964793 bytes from 1966080 bytes)
Building .pio/build/T-A7670X/firmware.bin
```

The firmware size exceeds the partition size. To get this fitting it's required to disable OTA or save some space.

naamah75, 2025-04-01:
It's okay not to use OTA. You can always flash the firmware using PlatformIO. Right?

closed 2025-04-03 by adlerre