adlerre/obd2-mqtt #57: Send MQTT before bluetooth connection?

naamah75, 2025-03-15:
It would be useful, at least for my case, if the MQTT connection was established even without having established a bluetooth connection to the ELM327, sending at predetermined intervals the diagnostics of the ESP32 peripheral (such as IP address, cellular signal strength, etc.) and GPS coordinates.

adlerre, 2025-03-16:
Currently the adapter mac address is used as identifier. So it's required to connect before sending data.

naamah75, 2025-03-16:
maybe can be user the esp32 mac instead?

adlerre, 2025-03-16:
This would work, but this would also change the identifier for all currently running installations.
Also a major rework of the code is required to send only this data.

adlerre, 2025-04-05:
I'm closing this because I want a BT/BLE connection to be established before sending data.

closed 2025-04-05 by adlerre