adlerre/obd2-mqtt #59: RTCWDT_RTC_RESET

naamah75, 2025-04-04:
I tried to install the new version 0.20.0 via web installer but I get a reset when the device boots.
As a device I selected T-A7670X_BLE, the same one I selected before but with _BLE

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46
```

adlerre, 2025-04-04:
Ups, my fault. Forgot to fix the offsets within manifest.

naamah75, 2025-04-04:
No, problem! :) please let me know when the modify is ready, so I can try  with web installer

naamah75, 2025-04-04:
just another little thing... sorry for bother you. Tried to compile with PlatformIO but I got this:

```
Esecuzione dell'attività: C:\Users\webma\.platformio\penv\Scripts\platformio.exe run 

Processing T-A7670X_BLE (board: esp32dev; platform: espressif32 @ ^6.10.0; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TinyGSM Fork @ 1.0.0+sha.ab40574
|-- ELMDuino @ 3.4.1
|-- PubSubClient @ 2.8.0
|-- ArduinoHttpClient @ 0.6.1
|-- ArduinoJson @ 7.3.1
|-- StreamDebugger @ 1.0.1
|-- AsyncTCP @ 3.3.2
|-- ESPAsyncWebServer @ 3.6.0
|-- ExprParser
|-- LittleFS @ 2.0.0
|-- WiFi @ 2.0.0
|-- MQTTWebSocketStreamClient
|-- FS @ 2.0.0
|-- Wire @ 2.0.0
|-- BLEClientSerial
|-- BluetoothSerial @ 2.0.0
|-- Update @ 2.0.0
Building in release mode
Compiling .pio\build\T-A7670X_BLE\src\OBDState.cpp.o
Compiling .pio\build\T-A7670X_BLE\src\OBDStates.cpp.o
Compiling .pio\build\T-A7670X_BLE\src\gsm.cpp.o
Compiling .pio\build\T-A7670X_BLE\src\helper.cpp.o
xtensa-esp32-elf-g++: error: unrecognized command line option '-q'
xtensa-esp32-elf-g++: error: unrecognized command line option '--short'; did you mean '--assert'?
xtensa-esp32-elf-g++: error: unrecognized command line option '--tags'; did you mean '-Wtabs'?
xtensa-esp32-elf-g++: error: unrecognized command line option '--exact-match)}\ -1 --pretty=%h)\'
xtensa-esp32-elf-g++: error: unrecognized command line option '-q'
xtensa-esp32-elf-g++: error: unrecognized command line option '--short'; did you mean '--assert'?
xtensa-esp32-elf-g++: error: unrecognized command line option '--tags'; did you mean '-Wtabs'?
xtensa-esp32-elf-g++: error: unrecognized command line option '--exact-match)}\ -1 --pretty=%h)\'
xtensa-esp32-elf-g++: error: unrecognized command line option '-q'
xtensa-esp32-elf-g++: error: unrecognized command line option '--short'; did you mean '--assert'?
xtensa-esp32-elf-g++: error: unrecognized command line option '--tags'; did you mean '-Wtabs'?
xtensa-esp32-elf-g++: error: unrecognized command line option '--exact-match)}\ -1 --pretty=%h)\'
Compiling .pio\build\T-A7670X_BLE\src\http.cpp.o
*** [.pio\build\T-A7670X_BLE\src\OBDStates.cpp.o] Error 1
*** [.pio\build\T-A7670X_BLE\src\gsm.cpp.o] Error 1
*** [.pio\build\T-A7670X_BLE\src\helper.cpp.o] Error 1
xtensa-esp32-elf-g++: error: unrecognized command line option '-q'
xtensa-esp32-elf-g++: error: unrecognized command line option '--short'; did you mean '--assert'?
xtensa-esp32-elf-g++: error: unrecognized command line option '--tags'; did you mean '-Wtabs'?
xtensa-esp32-elf-g++: error: unrecognized command line option '--exact-match)}\ -1 --pretty=%h)\'
*** [.pio\build\T-A7670X_BLE\src\OBDState.cpp.o] Error 1
xtensa-esp32-elf-g++: error: unrecognized command line option '-q'
xtensa-esp32-elf-g++: error: unrecognized command line option '--short'; did you mean '--assert'?
xtensa-esp32-elf-g++: error: unrecognized command line option '--tags'; did you mean '-Wtabs'?
xtensa-esp32-elf-g++: error: unrecognized command line option '--exact-match)}\ -1 --pretty=%h)\'
*** [.pio\build\T-A7670X_BLE\src\http.cpp.o] Error 1
```

adlerre, 2025-04-04:
The Flash issue with the web installer has been fixed.
I've never tested the build on Windows. However, there seem to be issues generating build flags (see platformio.ini, line 34-35).

naamah75, 2025-04-04:
Yes, I had seen that it was a problem with the Xtensa package for ESP32, but before 0.20.0 I was able to compile, maybe you could give me some tips. Can you fill it in? can I try with linux, what OS do you use?

adlerre, 2025-04-04:
Hm, you can try to run...

```bash
git symbolic-ref -q --short HEAD || git describe --tags --exact-match
git log -1 --pretty=%h
```
...and check the output.

Linux should be fine. I'm using Mac OS.

naamah75, 2025-04-04:
... anyway, the webinstaller works, in the next days I will try with the car. Thanks!

closed 2025-04-05 by adlerre