adlerre/obd2-mqtt #61: [Q] How to query custom PIDs

koostamas, 2025-04-05:
I have a VW MQB platform car. Unfortunately, VWs respond only to a very few standard PIDs. I am trying to add custom PIDs. For example, I had a custom PID added and working for the odometer in Torque (Android app) that looked like this:
- OBD2 Mode and PID: 2210e0
- Equation: (B*65536)+(C*256)+D
- ODB header: 7E0
I cannot figure out how I can input this PID into the service and PID fields or if it is even supported.
Do you have any idea?
Btw, I love the project.

adlerre, 2025-04-05:
> * OBD2 Mode and PID: 2210e0
> * Equation: (B * 65536) + (C * 256) + D
> * ODB header: 7E0

I think like so!

[states-2.json](https://github.com/user-attachments/files/19615076/states-2.json)

But it can be possible to need a scale factor like 1 / 10;

>Btw, I love the project.

Thank you i appreciate it!

> PS: If this is possible, I would like to create a new profile that supports the most useful VW MQB codes.

Yes of course, would be great!

koostamas, 2025-04-05:
Thank you for the quick response.
I tried your suggestion, but unfortunately it did not work. The result is still sent as 0.
Furthermore, I tried testing with [ELM327-emulator](https://github.com/Ircama/ELM327-emulator), and I see that the device sends  '2210E01' as command (note the extra '1' character at the end). Do you have any idea how I could get around this?
Btw, how did you come up with the '34' value for service? I could not figure that out.

adlerre, 2025-04-06:
You're absolutely right, the PID is being queried with "10E01."
That shouldn't happen. I need to analyze the ELMDuino code.

```
Service: 34
PID: 4320
Long query detected
Query string: 2210E01
Clearing input serial buffer
Sending the following command/query: 2210E01
        Received char: 4
        Received char: 1
        Received char: 0
        Received char: 0
        Received char: F
        Received char: F
        Received char: F
        Received char: F
        Received char: F
        Received char: F
        Received char: F
        Received char: F
        Received char: \r
        Received char: \r
        Received char: >
```

The service and PID values ​​are decimal instead of hexadecimal, so 0x22 corresponds to 34 decimal.

koostamas, 2025-04-06:
I see that the extra '1' is just the number of responses variable appended to the command here: [ELMduino repo](https://github.com/PowerBroker2/ELMduino/blob/e7c9e7b634876778f236a0eb544ea9fd5641ce14/src/ELMduino.cpp#L253C17-L253C56)
This same logic is applied to the normal queries as well, like for the standard odometer service and PID (1 and 166) it gets converted to '01A61' which is the correct command.
So it looks like that for this type of query we might not need to add the number of responses?
I will try to create a build myself later today where I change the linked line to not add the value and test with that.

adlerre, 2025-04-06:
I come to the same conclusion, but I think this is necessary to add the number of responses to the query. If this is zero, nothing is send back.

adlerre, 2025-04-06:
You don't need to change the ELMduino code, if you add...

```
elm327.specifyNumResponses = false;
```

after line 629 within obd.cpp, this "1" isn't added.

koostamas, 2025-04-06:
I see that the [competition](https://github.com/meatpiHQ/wican-fw/issues/108#issuecomment-2127817637) has this mostly figured out.

adlerre, 2025-04-07:
> Is there some way I can enable verbose logging about the ELM communication? Maybe there are some clues there.

You can enable debug logs on ELM327 init (second parameter on line 584/592) and send me the result.

```cpp
...
while (!elm327.begin(serialBLE, true, 2000, protocol) && retryCount < 3) {
...
```

koostamas, 2025-04-07:
Btw, this is my testing states.json:
```
[
  {
    "type": 1,
    "valueType": "int",
    "enabled": true,
    "visible": false,
    "interval": -1,
    "name": "startTime",
    "description": "Start Time",
    "measurement": false,
    "diagnostic": true,
    "expr": "($millis)",
    "pid": {
      "service": 0,
      "pid": 0,
      "numResponses": 0,
      "numExpectedBytes": 0,
      "bias": 0
    },
    "value": { "format": "%d" }
  },
  {
    "type": 0,
    "valueType": "int",
    "enabled": true,
    "visible": true,
    "interval": -1,
    "name": "supportedPids_1_20",
    "description": "Supported PIDs 1-20",
    "measurement": false,
    "diagnostic": true,
    "pid": {
      "service": 1,
      "pid": 0,
      "numResponses": 1,
      "numExpectedBytes": 4,
      "scaleFactor": "1",
      "bias": 0
    },
    "value": { "func": "toBitStr" }
  },
  {
    "type": 0,
    "valueType": "int",
    "enabled": true,
    "visible": true,
    "interval": -1,
    "name": "supportedPids_21_40",
    "description": "Supported PIDs 21-40",
    "measurement": false,
    "diagnostic": true,
    "pid": {
      "service": 1,
      "pid": 32,
      "numResponses": 1,
      "numExpectedBytes": 4,
      "scaleFactor": "1",
      "bias": 0
    },
    "value": { "func": "toBitStr" }
  },
  {
    "type": 0,
    "valueType": "int",
    "enabled": true,
    "visible": true,
    "interval": -1,
    "name": "supportedPids_41_60",
    "description": "Supported PIDs 41-60",
    "measurement": false,
    "diagnostic": true,
    "pid": {
      "service": 1,
      "pid": 64,
      "numResponses": 1,
      "numExpectedBytes": 4,
      "scaleFactor": "1",
      "bias": 0
    },
    "value": { "func": "toBitStr" }
  },
  {
    "type": 0,
    "valueType": "int",
    "enabled": true,
    "visible": true,
    "interval": -1,
    "name": "supportedPids_61_80",
    "description": "Supported PIDs 61-80",
    "measurement": false,
    "diagnostic": true,
    "pid": {
      "service": 1,
      "pid": 96,
      "numResponses": 1,
      "numExpectedBytes": 4,
      "scaleFactor": "1",
      "bias": 0
    },
    "value": { "func": "toBitStr" }
  },
  {
    "type": 0,
    "valueType": "int",
    "enabled": true,
    "visible": true,
    "interval": 100,
    "name": "odometer_custom",
    "description": "Odometer custom",
    "icon": "counter",
    "unit": "km",
    "deviceClass": "distance",
    "measurement": true,
    "diagnostic": false,
    "pid": {
      "service": 34,
      "pid": 4320,
      "numResponses": 2,
      "numExpectedBytes": 8,
      "scaleFactor": "1",
      "bias": 0
    },
    "value": { "format": "%d" }
  },
  {
    "type": 0,
    "valueType": "int",
    "enabled": true,
    "visible": true,
    "interval": 100,
    "name": "oil_level",
    "description": "Oil level",
    "icon": "counter",
    "unit": "mm",
    "deviceClass": "distance",
    "measurement": true,
    "diagnostic": false,
    "pid": {
      "service": 34,
      "pid": 4538,
      "numResponses": 1,
      "numExpectedBytes": 4,
      "scaleFactor": "1",
      "bias": 0
    },
    "value": { "format": "%d" }
  },
  {
    "type": 0,
    "valueType": "int",
    "enabled": true,
    "visible": true,
    "interval": 100,
    "name": "rpm",
    "description": "Revolutions per minute",
    "icon": "engine",
    "measurement": true,
    "diagnostic": false,
    "pid": {
      "service": 1,
      "pid": 12,
      "numResponses": 1,
      "numExpectedBytes": 2,
      "scaleFactor": "1.0 / 4.0",
      "bias": 0
    },
    "value": { "format": "%d" }
  }
]
```
And from the log these two queries seem weird to me:
```
[09:56:45]Service: 34
[09:56:45]PID: 160
```
```
[09:56:45]Service: 34
[09:56:45]PID: 224
```
Are these trying to test if the adapter supports the PID? Or just some kind of bug?

adlerre, 2025-04-07:
> Are these trying to test if the adapter supports the PID? Or just some kind of bug?

Yes, it is.

> I see that you added the existing VW e-UP profile in #25. How did you come up with the values required there? It uses the same "VW-specific" service, so I guess the same type of request should work for me, too.

I use the infos from here. But I can't confirm that this work.

adlerre, 2025-04-07:
Can you test the PR and post the ELM debug log? It's now possible to set the required extra header.

```
{
    "type": 0,
    "valueType": "int",
    "enabled": true,
    "visible": true,
    "interval": 100,
    "name": "odometer_custom",
    "description": "Odometer custom",
    "icon": "counter",
    "unit": "km",
    "deviceClass": "distance",
    "measurement": true,
    "diagnostic": false,
    "pid": {
      "service": 34,
      "pid": 4320,
      "header": 2016,
      "numResponses": 1,
      "numExpectedBytes": 4,
      "scaleFactor": "1",
      "bias": 0
    },
    "value": { "format": "%d" }
  },
```

koostamas, 2025-04-07:
Success! It worked.

Here are the logs:
```
[19:03:06]Sending the following command/query: AT D
[19:03:06]	Received char: A
[19:03:06]	Received char: T
[19:03:06]	Received char: _
[19:03:06]	Received char: D
[19:03:06]	Received char: \r
[19:03:06]	Received char:  
[19:03:06]	Received char: O
[19:03:06]	Received char: K
[19:03:06]	Received char: \r
[19:03:06]	Received char: \r
[19:03:06]	Received char: >
[19:03:06]Delimiter found.
[19:03:07]Clearing input serial buffer
[19:03:07]Sending the following command/query: AT Z
[19:03:07]	Received char: A
[19:03:07]	Received char: T
[19:03:07]	Received char: _
[19:03:07]	Received char: Z
[19:03:07]	Received char: \r
[19:03:07]	Received char:  
[19:03:07]	Received char: \r
[19:03:07]	Received char: \r
[19:03:07]	Received char: E
[19:03:07]	Received char: L
[19:03:07]	Received char: M
[19:03:07]	Received char: 3
[19:03:07]	Received char: 2
[19:03:07]	Received char: 7
[19:03:07]	Received char: _
[19:03:07]	Received char: v
[19:03:07]	Received char: 1
[19:03:07]	Received char: .
[19:03:07]	Received char: 5
[19:03:07]	Received char: \r
[19:03:07]	Received char: \r
[19:03:07]	Received char: >
[19:03:07]Delimiter found.
[19:03:07]Clearing input serial buffer
[19:03:07]Sending the following command/query: AT E0
[19:03:08]	Received char: A
[19:03:08]	Received char: T
[19:03:08]	Received char: _
[19:03:08]	Received char: E
[19:03:08]	Received char: 0
[19:03:08]	Received char: \r
[19:03:08]	Received char: O
[19:03:08]	Received char: K
[19:03:08]	Received char: \r
[19:03:08]	Received char: \r
[19:03:08]	Received char: >
[19:03:08]Delimiter found.
[19:03:08]Clearing input serial buffer
[19:03:08]Sending the following command/query: AT S0
[19:03:08]	Received char: O
[19:03:08]	Received char: K
[19:03:08]	Received char: \r
[19:03:08]	Received char: \r
[19:03:08]	Received char: >
[19:03:08]Delimiter found.
[19:03:08]Clearing input serial buffer
[19:03:08]Sending the following command/query: AT AL
[19:03:08]	Received char: O
[19:03:08]	Received char: K
[19:03:08]	Received char: \r
[19:03:08]	Received char: \r
[... 1370 lines cut ...]
[19:03:57]Clearing input serial buffer
[19:03:57]Sending the following command/query: 010C
[19:03:57]	Received char: 0
[19:03:57]	Received char: 1
[19:03:57]	Received char: 0
[19:03:57]	Received char: C
[19:03:57]	Received char: \r
[19:03:57]	Received char: S
[19:03:57]	Received char: E
[19:03:57]	Received char: A
[19:03:57]	Received char: R
[19:03:57]	Received char: C
[19:03:57]	Received char: H
[19:03:57]	Received char: I
[19:03:57]	Received char: N
[19:03:57]	Received char: G
[19:03:57]	Received char: .
[19:03:57]	Received char: .
[19:03:57]	Received char: .
[19:03:57]	Received char: \r
```

For a second try, I also enabled an oil level pid with the same header (the one you can see in my states above), but that also had the same value as the odometer. Kind of weird, but I do not have the logs for that and didn't investigate, this is enough progress for today, I will check again tomorrow.

I am very grateful for your help!

adlerre, 2025-04-08:
> Success! It worked.

Does that mean the correct mileage was sent?

> I am very grateful for your help!

Me too, because all of this makes the project better.

koostamas, 2025-04-08:
> Does that mean the correct mileage was sent?

Yep

koostamas, 2025-04-08:
My next question for this: The sensor I tried to add yesterday (oil level) requires to execute an equation on the response to get the correct value, in this case it's `(A*256+B)/249.6`. Where can I input this?

closed 2025-04-08 by adlerre