adlerre/obd2-mqtt #74: How to use CALC function with signed first byte?

ivobacco, 2025-05-22:
Hello, at first thanks for the great project. I've been using it since a few weeks with my Mustang Mach-E EV. 
I've got a few OBD PIDs that requires a calculation to get the correct value, but so far I'm struggling in understanding how to do it. 

Example:

| Name                          | ShortName  | HA  | ModeAndPID | PidDec | Equation                                                                                                | Min Value | Max Value | Units     | Header |
| ----------------------------- | ---------- | --- | ---------- | ------ | ------------------------------------------------------------------------------------------------------- | --------- | --------- | --------- | ------ |
| HVB Current                   | HvbA       | YES | 0x2248f9   | 18681  | ((signed(A)\*256)+B)\*0.1                                                                               | \-600     | 600       | Amps      | 7e4    |
| HVB Power Flow Calculated     | HvbPwr     | YES |            |        | val{HVB Current}\*val{HVB Voltage}\*0.001                                                               | \-200     | 200       | kW        |        |
| HVB Voltage                   | HvbV       | YES | 0x22480d   | 18445  | INT16(A:B)\*0.01                                                                                        | 0         | 500       | Volts     | 7e4    |
| Instant Wh/km               | InWhMi     | YES |            |        | val{HvbPwr}/val{VSSHR}\*1000                                                                            | \-600     | 600       | Wh/km     |        |
| Vehicle Speed High Resolution | VSSHR      | YES | 0x221505   | 5381   | INT16(A:B)/128                                                                                | 0         | 12000     | KPH | 7e0    |

Two questions: 
 - how do I calculate correct value for PID of "HVB Current"? 
 - how do I calculate value for "HVB Power Flow" when "HVB voltage" is greater than one single byte? 

Thanks in advance. 
 Ivo

adlerre, 2025-05-23:
Hello, thanks!

Sorry for the late reply. I created a small example of these values ​​and found an error in the form. You can add it to your current JSON. I hope it works as expected.

```json
[
  {
    "type": 0,
    "valueType": "float",
    "enabled": true,
    "visible": true,
    "interval": 100,
    "name": "HvbA",
    "description": "HVB Current",
    "icon": "current-ac",
    "unit": "A",
    "deviceClass": "current",
    "measurement": true,
    "diagnostic": false,
    "pid": {
      "service": 34,
      "pid": 18681,
      "header": 2020,
      "numResponses": 1,
      "numExpectedBytes": 2,
      "scaleFactor": "1.0 / 10.0",
      "bias": -600
    }
  },
  {
    "type": 0,
    "valueType": "float",
    "enabled": true,
    "visible": true,
    "interval": 100,
    "name": "HvbV",
    "description": "HVB Voltage",
    "icon": "flash-triangle",
    "unit": "V",
    "deviceClass": "voltage",
    "measurement": true,
    "diagnostic": false,
    "pid": {
      "service": 34,
      "pid": 18445,
      "header": 2020,
      "numResponses": 1,
      "numExpectedBytes": 2,
      "scaleFactor": "1.0 / 100.0",
      "bias": 0
    }
  },
  {
    "type": "1",
    "valueType": "float",
    "enabled": true,
    "visible": true,
    "interval": 100,
    "name": "HvbPwr",
    "description": "HVB Power Flow Calculated",
    "icon": "power-plug",
    "unit": "kW",
    "deviceClass": "power",
    "measurement": false,
    "diagnostic": false,
    "expr": "$HvbA * $HvbV * 0.001"
  }
]
```

René

adlerre, 2025-05-25:
Can you provide some logs with enabled debug? It's easier to understand in bytes. ;) It's seams a number type conversion is required.

adlerre, 2025-05-28:
Thanks for the Log. Allow me some time to investigate and develop a solution.

ivobacco, 2025-06-01:
Hello René,
I'm not sure how you are doing that conversions, I got inspiration from the awesome work done here by [sillygoose](https://github.com/sillygoose/mme-record), calculations [here](https://github.com/sillygoose/mme-record/blob/3bfad3763d598132b7b4db8bf7425794d3826f76/source/codec_manager.py) and I wrote I small python script which uses PID 18681 and 18445 to calculate "HVB Power Flow Calculated".

In the [script_pastebin](https://pastebin.com/CznnPDBj) I used the PID values extracted from the log file I linked above. 

HVB current: 38.7 A |  HVB voltage: 379.98 V |  HVB Power Flow Calculated: 14.71 kW
HVB current: 11.5 A |  HVB voltage: 380.60 V |  HVB Power Flow Calculated: 4.38 kW
HVB current: 9.5 A |  HVB voltage: 380.23 V |  HVB Power Flow Calculated: 3.61 kW
HVB current: 15.0 A |  HVB voltage: 381.33 V |  HVB Power Flow Calculated: 5.72 kW
HVB current: -25.9 A |  HVB voltage: 381.33 V |  HVB Power Flow Calculated: -9.88 kW
HVB current: 1.6 A |  HVB voltage: 380.72 V |  HVB Power Flow Calculated: 0.61 kW
HVB current: 1.1 A |  HVB voltage: 380.72 V |  HVB Power Flow Calculated: 0.42 kW
HVB current: 1.1 A |  HVB voltage: 380.47 V |  HVB Power Flow Calculated: 0.42 kW
HVB current: 1.1 A |  HVB voltage: 380.60 V |  HVB Power Flow Calculated: 0.42 kW
HVB current: 1.0 A |  HVB voltage: 380.60 V |  HVB Power Flow Calculated: 0.38 kW
HVB current: 1.0 A |  HVB voltage: 380.72 V |  HVB Power Flow Calculated: 0.38 kW
HVB current: 1.1 A |  HVB voltage: 380.72 V |  HVB Power Flow Calculated: 0.42 kW
HVB current: 1.0 A |  HVB voltage: 380.60 V |  HVB Power Flow Calculated: 0.38 kW
HVB current: -0.6 A |  HVB voltage: 380.47 V |  HVB Power Flow Calculated: -0.23 kW
HVB current: 24.6 A |  HVB voltage: 379.62 V |  HVB Power Flow Calculated: 9.34 kW
HVB current: 1.3 A |  HVB voltage: 380.96 V |  HVB Power Flow Calculated: 0.50 kW
HVB current: 4.3 A |  HVB voltage: 380.47 V |  HVB Power Flow Calculated: 1.64 kW
HVB current: 11.9 A |  HVB voltage: 380.11 V |  HVB Power Flow Calculated: 4.52 kW
HVB current: 27.5 A |  HVB voltage: 379.62 V |  HVB Power Flow Calculated: 10.44 kW
HVB current: -4.3 A |  HVB voltage: 380.96 V |  HVB Power Flow Calculated: -1.64 kW
HVB current: 1.6 A |  HVB voltage: 380.60 V |  HVB Power Flow Calculated: 0.61 kW
HVB current: 1.1 A |  HVB voltage: 380.47 V |  HVB Power Flow Calculated: 0.42 kW
HVB current: 1.2 A |  HVB voltage: 380.47 V |  HVB Power Flow Calculated: 0.46 kW
HVB current: 1.2 A |  HVB voltage: 380.47 V |  HVB Power Flow Calculated: 0.46 kW
HVB current: 2.8 A |  HVB voltage: 380.60 V |  HVB Power Flow Calculated: 1.07 kW
HVB current: 1.0 A |  HVB voltage: 380.60 V |  HVB Power Flow Calculated: 0.38 kW
HVB current: -28.0 A |  HVB voltage: 381.57 V |  HVB Power Flow Calculated: -10.68 kW
HVB current: 126.5 A |  HVB voltage: 375.71 V |  HVB Power Flow Calculated: 47.53 kW
HVB current: 11.9 A |  HVB voltage: 380.11 V |  HVB Power Flow Calculated: 4.52 kW

The calculation of the hexadecimal conversion matches with web calculators. example [here](https://www.rapidtables.com/convert/number/hex-to-decimal.html?x=0183) under "signed 2 compliment". 

I'm quite sure the PID number is correct for my model of Mach-E. 

Thanks!
 Ivo

adlerre, 2025-06-01:
What firmware are you using?
I add a additional option to set the right response format.

adlerre, 2025-06-01:
I would be nice if you can test this firmware. I added an option to define the response format. For PID 18681 you can select INT_16.

[OBD2-MQTT-T-A7670X-response-format_(12513b3).zip](https://github.com/user-attachments/files/20541128/OBD2-MQTT-T-A7670X-response-format_.12513b3.zip)

ivobacco, 2025-06-05:
Hello René, works! 
I finally had the chance to test it, values for 18681 PID now looks in range. 

HVB current: 1.9 A |  HVB voltage: 367.64 V |  HVB Power Flow Calculated: 0.70 kW
HVB current: 54.7 A |  HVB voltage: 365.44 V |  HVB Power Flow Calculated: 19.99 kW
HVB current: 4.7 A |  HVB voltage: 367.40 V |  HVB Power Flow Calculated: 1.73 kW
HVB current: 48.0 A |  HVB voltage: 363.61 V |  HVB Power Flow Calculated: 17.45 kW
HVB current: -45.6 A |  HVB voltage: 368.37 V |  HVB Power Flow Calculated: -16.80 kW
HVB current: 10.8 A |  HVB voltage: 367.52 V |  HVB Power Flow Calculated: 3.97 kW
HVB current: 3.0 A |  HVB voltage: 367.40 V |  HVB Power Flow Calculated: 1.10 kW
HVB current: 71.5 A |  HVB voltage: 364.22 V |  HVB Power Flow Calculated: 26.04 kW
HVB current: 54.8 A |  HVB voltage: 365.44 V |  HVB Power Flow Calculated: 20.03 kW
HVB current: -117.9 A |  HVB voltage: 369.47 V |  HVB Power Flow Calculated: -43.56 kW
HVB current: 2.6 A |  HVB voltage: 367.52 V |  HVB Power Flow Calculated: 0.96 kW
HVB current: -22.5 A |  HVB voltage: 367.64 V |  HVB Power Flow Calculated: -8.27 kW
HVB current: -30.7 A |  HVB voltage: 367.64 V |  HVB Power Flow Calculated: -11.29 kW

[image]

Would you mind adding ENERGY_DISTANCE and ENERGY_STORAGE device classes along with their measurement? 
Wish you a very nice day.

closed 2025-06-05 by adlerre