adlerre/obd2-mqtt #76: Send static diagnostic data failed

The-Noah, 2025-05-31:
I have a SIM800L and got it all connected, but no data is in Home Assistant and my logs are filled with `Send static diagnostic data......failed`.

```
[13:13:15]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:13:15]configsip: 0, SPIWP:0xee
[13:13:15]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:13:15]mode:DIO, clock div:2
[13:13:15]load:0x3fff0030,len:1184
[13:13:15]load:0x40078000,len:13232
[13:13:15]load:0x40080400,len:3028
[13:13:15]entry 0x400805e4
[13:13:15]E (609) esp_core_dump_flash: No core du��partition found!
[13:13:15]E (609) esp_core_dump_flash: No core dump partition found!
[13:13:16]Start Access Point...WiFi AP started.
[13:13:16]AP - IP address: 192.168.4.1
[13:13:16]WiFi AP stopped.
[13:13:16]WiFi AP started.
[13:13:16]AP - IP address: 192.168.4.1
[13:13:16]HTTP server started
[13:13:16]Start modem...
[13:13:20]Initializing modem...
[13:13:22]Modem Name: SIMCOM SIM800L
[13:13:22]Modem Info: SIM800 R14.18
[13:13:27]Waiting for network......success
[13:13:27]Network connected
[13:13:34]Connecting to super......success
[13:13:34]GPRS connected
[13:13:34]IP Address: 100.69.236.172
[13:13:36]connecting to 1c:a1:35:69:8d:c5
[13:13:37]Connected to ELM327
[13:13:37]ELM327 protocol: ATDP
[13:13:42]Read location......done (5089ms)
[13:13:48]The client obd2mqtt-1ca135698dc5 connects to the MQTT broker......connected.
[13:13:58]Send discovery data......done (9823ms)
[13:13:59]Send diagnostic discovery data......done (1232ms)
[13:14:00]Send static diagnostic discovery data......done (633ms)
[13:14:00]Send diagnostic data......done (239ms)
[13:14:00]Send static diagnostic data......failed (0ms)
[13:14:03]Read location......done (2788ms)
[13:14:06]The client obd2mqtt-1ca135698dc5 connects to the MQTT broker......connected.
[13:14:06]Send static diagnostic data......failed (0ms)
[13:14:06]Send static diagnostic data......failed (0ms)
[13:14:06]Send static diagnostic data......failed (0ms)
[13:14:07]Send static diagnostic data......failed (0ms)
[13:14:07]Send static diagnostic data......failed (0ms)
[13:14:07]Send static diagnostic data......failed (0ms)
[13:14:07]Send static diagnostic data......failed (0ms)
[13:14:07]Send static diagnostic data......failed (1ms)
[13:14:07]Send static diagnostic data......failed (0ms)
[13:14:07]Send static diagnostic data......failed (1ms)
[13:14:07]Send static diagnostic data......failed (1ms)
[13:14:07]Send static diagnostic data......failed (0ms)
[13:14:07]Send static diagnostic data......failed (0ms)
[13:14:07]Send static diagnostic data......failed (1ms)
[13:14:07]Send static diagnostic data......failed (0ms)
[13:14:07]Send static diagnostic data......failed (0ms)
[13:14:07]Send static diagnostic data......failed (1ms)
[13:14:07]Send static diagnostic data......failed (0ms)
[13:14:07]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (1ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (1ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:08]Send static diagnostic data......failed (0ms)
[13:14:09]Send static diagnostic data......failed (0ms)
[13:14:09]Send static diagnostic data......failed (0ms)
[13:14:09]Send static diagnostic data......failed (1ms)
[13:14:09]Send static diagnostic data......failed (0ms)
[13:14:09]Send static diagnostic data......failed (1ms)
[13:14:09]Send static diagnostic data......failed (0ms)
[13:14:09]Send static diagnostic data......failed (0ms)
[13:14:09]Send static diagnostic data......failed (0ms)
[13:14:09]Send static diagnostic data......failed (0ms)
```

... and it goes on and on and on

adlerre, 2025-06-01:
Hi, if tested this with mine without any problems. Which version you use, have you ticked "check if PID supported"?

The-Noah, 2025-06-02:
I have a [LILYGO® TTGO T-Call V1.4 SIM800L](https://www.aliexpress.us/item/2251832858907208.html?gatewayAdapt=glo2usa) with the `SIM800L_IP5306_VERSION_20200811_BLE` firmware. I did not have "check if PID supported" enabled, so I enabled it and still same result.

The-Noah, 2025-06-05:
Whenever I have any of the supported PIDs diagnostic enabled I get the error. When I disable all 3 of them, I only see this in Home Assistant:

[image]