adlerre/obd2-mqtt #81: Help with the cheap adapter

mrogers-medicareinsurance, 2025-08-17:
Hi, in the readme it says "if you want to use [this](https://de.aliexpress.com/item/1005005775562398.html) cheap adapter, you should set the right connection protocol or have patience"

Can you elaborate on this? I'm trying to use your code [here](https://github.com/vdvornichenko/obd-ble-serial/pull/1/commits/be029ae1282b962cde592b9c920846d6cb6562f9) to have ESP32-S3 talk to the cheap blue ELM327 and I can't get it to work

adlerre, 2025-08-17:
Hi,

an ESP-S3 can only used with a BLE adapter, Bluetooth isn‘t supported by this device.

mrogers-medicareinsurance, 2025-08-17:
I don't totally understand. ESP32-S3 (In my case Arduino Nano ESP32) only supports BLE, and those blue ELM327 are classic bluetooth only and can't use BLE at all? So that library would only help for using ELMduino with a different OBD device that is specifically BLE?

adlerre, 2025-08-17:
Yes, you need a BLE-Adapter like [this](https://www.amazon.de/dp/B0D474RC4W?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_1&th=1).
However, this project does not support the Arduino Nano ESP32, but requires an ESP with a GSM module. 
If you are looking for a library to support ELMDuino, use [BLESerial](https://github.com/adlerre/BLESerial).

closed 2025-10-01 by github-actions[bot]